LaserWeb/lw.comm-server #109: Unsupported protocol version

cpthowdy666, 2022-06-18:
Currently running the server on a Jammy Jellyfish SBC, followed all steps provided in guide.
When I try to connect to the server, I get `code: 5, message: Unsupported protocol version` response for all web socket calls.
Server version 4.1.000
Front end version: 4.0.999 (the most current available version for windows).

LaserWeb works great when run locally, but that's not viable for my setup.

I saw there was recently a 4.1.000 version for Mac - is this just something to wait out?

cpthowdy666, 2022-06-18:
After playing with a few things in server.js - changing the websockets constructor to have allowEIO3 set to true and cors.origin and cors.credentials set to true, I'm able to at least connect to the server. However, at this point I'm not getting the list of USBs from the server to use.

cprezzi, 2022-06-18:
Sounds like missmatched versions of socket.io (or our protocol on top) between backend and frontend. 
@easytarget is working on a updated frontend which needs some adjustments on the socket communication. If you combine the wrong versions, is will not work.

Did you try to start the frontend via Webbrowser (backend IP:8000)? This way you should get matching versions.

cpthowdy666, 2022-06-18:
Thanks for the reply @cprezzi 

I was able to start it that way, yes. I dug into the issues after I posted this one and saw that work around. Had to fight some board issues, but we look to be up and working now.

closed 2022-06-18 by cpthowdy666