LaserWeb/lw.comm-server #112: Failed at the node-hid@2.1.2 install script

miserovitch, 2023-01-03:
Hi,

I am trying to install laserweb on  a PI3B with a fresh install:
bullseye V11
node 12.22.12
npm 6.14.16

I did the apt update and upgrade ( i tried also without) and I have an issue when installing npm in lw.comm-server folder.
I think it is related to node-hid install.

`3954 error node-hid@2.1.2 install: `prebuild-install --runtime napi || node-gyp rebuild`

3954 error Exit status 1`

I enclosed the log.

Anyone able to support on this issue?

Thks

[2023-01-03T09_32_21_063Z-debug.log](https://github.com/LaserWeb/lw.comm-server/files/10335823/2023-01-03T09_32_21_063Z-debug.log)

cprezzi, 2023-01-03:
The node-hid library is only needed when you connect a USB/Wireless CNC Remote Control to the server. Try to delete the corresponding line in package.json and run "npm install" again.

cprezzi, 2023-01-03:
Eventualy there is a incompatibility in node-hid 2.1.2.
Please try to remove the ^ in the node-hid line in package.json to force the use of v2.1.1.

miserovitch, 2023-01-04:
Hi,
Thks @cprezzi !

I removed the ^ for node-hid in json file and it worked.

closed 2023-01-04 by miserovitch

cprezzi, 2023-01-04:
Thank you for the reply.